pgrady3/EasyController3 #2: Mapping of current and voltage scaling

krishna-1728, 2023-06-08:
We have recently encountered a situation where the board's response to throttle input is unresponsive, with voltage readings remaining stagnant at an oscillation between 0 and 1 volts even when full throttle is applied. After thorough analysis, it has become apparent that there might be a slight discrepancy in the current scaling and voltage synchronization with our motor of choice or there might be an issue with the calibration of the board and motor. It would be helpful if you could suggest a few measures to overcome the issue.

pgrady3, 2023-07-07:
Hey Krishna, that description is helpful, but not quite detailed enough for me to know what's going on. Are you talking about the current control loop?

You can try to manually set the duty cycle and that will always make the motor go.

krishna-1728, 2023-07-07:
The motor is not responding to any input from the throttle because of which it is static. We are not able to figure out what the issue is in the controller. Can you suggest a way to make it work? 
What are all things I should consider in the motor controller and make relevant changes to it?
In the previous question I am talking about the output we are getting at all three phases in the motor depending upon the sequence. The range of the output is 0 -1 volts.

pgrady3, 2023-07-07:
Is the controller correctly reading the throttle input from the ADC? You could try printing out the ADC value for the throttle pin.

You can also try attaching a video or photos to give a better sense of what's going on.

krishna-1728, 2023-07-07:
The controller is correctly reading the ADC value from the throttle pin, I have checked it . 
Initially, all three phases show 9.8 volts as soon as I give the throttle two phases become zero and the third phase shows variation from 0 -1 volts depending upon the throttle input.

pgrady3, 2023-07-07:
There is a chance the gate drivers are not working properly. I would look to see what duty cycle the controller is outputting, and verify with an oscilloscope if the microcontroller is correctly doing that, and if the gate drivers are correctly doing what they're told.

Sorry, I'm not sure that I can be more of a help without more info.

krishna-1728, 2023-07-07:
Can you please provide me with more detail about the duty cycle it should give, so that I can check that on oscilloscope?
I have created one more speed controller but I am facing issues with the duty cycle and frequency. 
 I have attached a copy of the simple code which I have written in Python to control the controller. If you could suggest any changes ? 

[Python_code.docx](https://github.com/pgrady3/EasyController3/files/11984897/Python_code.docx)