pgrady3/EasyController3 #3: Regenerative Braking

pirdain, 2023-07-07:
Hey, it says in the Readme that Regenerative Braking was added in the EasyController3 and is an upgrade over the EasyController2, but I'm unable to find any support in the Hardware or Software regarding it. I believe an external circuit would be added to support regenerative braking and supplying current back to the battery pack at the required voltage and this may or may not require some additions in the software (for the basic functionality at least).

If anyone has figured out how to incorporate Regenerative Braking into this project please let me know.

Cheers.

pgrady3, 2023-07-07:
Hi Pirdain,

It's just a software change, and if you set the desired current value negative, the control loop will modulate the duty cycle to make current will flow backwards into the battery. No hardware change is required. However, I haven't tested this yet.

Sorry for the wait, I'm working on the updated 3.1 version that will have this feature enabled and tested.

Patrick

pirdain, 2023-07-10:
Hey Patrick,

Thanks for the quick response. I'll test this with my motor and get back with the results.

Also I had another question, how do I calculate the Capacitance value for the Capacitor Bank. Did you follow some formula for it. The provided value is as per a lower wattage motor, 1 kW. The motor in my circuit is around 5 - 6 kW.

I understand that the Issue is for Regenerative Braking and this would need another issue altogether, but still any input is appreciated.

Kind Regards,
Pirdain

pgrady3, 2023-07-10:
Hey Pirdain,

Choosing these bypass capacitors is a very empirical process. I'd recommend finding a comparable 5 kW controller, and just copying the capacitance value from that. Alternatively, you could design in 5x more capacitance, and solder in different values depending on what you find when testing.

Patrick

bmentink, 2024-04-09:
Just a point regarding managing regen energy. If the current limit to the battery is set too low, or the battery won't accept any more current (fully charged), the regen voltage may still rise to a damaging level. One way around that is to dump to an external high power resistor in that situation. 

Maybe that could be added to version4 ;)

pgrady3, 2024-04-11:
Yes, some controllers have a shunt resistor for these kinds of applications. I think this is mostly used for industrial BLDC servos, where dumping energy back into the bus is unacceptable.

On most vehicle applications (what this controller is really built for), you apply the mechanical brakes if the controller can't regen. And usually, the battery's stored energy is much larger than the kinetic energy of the vehicle, so over-charging the battery isn't a concern.

However, you're welcome to add more features to the design!

closed 2024-04-11 by pgrady3

bmentink, 2024-04-19:
Hi Patrick,

I am making great progress with sensorless addition to easycontroller3. The
zero-cross detection is achieved with simple (two resistors and a fet)
components  per phase. I have used a majority filter function in software
to clean it up, works great between 18v and 48v, no hardware filtering at
all, just straight into the ADC.

Attached is a scope shot.

Cheers,
Bernard

bmentink, 2024-05-17:
Hi Patrick,

I have finally managed to get an implementation of sensorless operation
working on your board. It works reliably over a voltage range of 12v to 48v
with no hardware changes needed to support different voltages, I can just
ramp up the bench power supply to whatever voltage. And it starts up
reliably , even under load.

I can't stop the motor by loading the shaft with my hand, but will need
real load testing.

The hardware involved is 3 small signal fet',s a bunch of resistors and the
CD4051 mux .. (finally got that working correctly) so would
not be much to add to the bottom of the board.

At the moment, for development, I have totally re-written your software to
support this mode, and I have yet to add back current/torque control. (Also
it is written in Forth, not C, because it was fast to test out code for
development) When I am totally happy with it I could port it back to C ...

Just thought I would keep you up to date  with the development.

Cheers,
Bernie