pgrady3/EasyController3 #4: Version 3.1

samuk, 2023-08-24:
Is the new version still planned?

krekr, 2023-10-01:
I’m watching this also. If the fix can be described, that would help as well. Thanks!

samuk, 2023-10-01:
Yes that might help. @pgrady3 could you describe the problem?

TinkerGear, 2023-10-08:
A description of the problem would be very helpful to us as well!

pgrady3, 2023-10-09:
Hi all, apologies for the slow reply. I'm pretty busy here finishing up with my PhD, hence the slow progress. Adding a 22 ohm resistor between the transistor gate and the gate drivers fixes the problem of the gate drivers blowing up.

However, there were a few other issues that I've also been working on:

- The components are a bit too hard to solder, and determining the orientation of the SOT-23-6's is almost impossible. I'm switching them out for bigger components
- The code is a mess. It needs to be cleaned up and 
- I think the gate driver deadtime is too short, and the transistors heat up a too much during operation.

Apologies on the buggy release. I rushed this one, and as a result it's not very polished.

krekr, 2023-10-09:
Thanks very much. This gives me enough confidence to fork it and design a smaller version that fits into a certain application.

samuk, 2023-10-16:
@krekr I'd be interested in checking out your fork if you could post it here.

krekr, 2024-02-23:
Huzzah for v3.2! Thanks @pgrady3

closed 2024-02-23 by samuk