pgrady3/EasyController3 #9: BLDC Hall sensors output 5v - should this be divided for the pi pico?

joshmcculloch, 2024-05-30:
Hello,

I have been doing some testing and the driver is preforming well. I was a bit concerned to see 5v on the hall input pins. Do we need to add a resistor divider to the design. Or is it unusual for the hall logic output to be 5v? I see some of the hall sensors support 3.3-5v, so another option might be to power them from the pico's 3v3 rail. 

Also happy to hear that I'm wrong and this is a non issue.

joshmcculloch, 2024-05-30:
Follow up. It seems that hall effects are generally open collector - mine are. So the solution I'm going to try is breaking the 5v pull up line along the edge of the board and connecting it to the pi's 3v3 via the R3 pad.

l4m4re, 2024-07-15:
In my design (still in progress) for a controller with an asymmetric bridge, I opted for ISO7760FQDWRQ1 isolated buffers:

https://github.com/l4m4re/BEMF_Controller/blob/main/kicad/BEMF_controller.pdf

[image]