pgrady3/EasyController3 #10: My version of the controller that is now sensorless ..

bmentink, 2024-06-23:
Did a version of the board (in Kicad) that does sensorless, all that was added is 3 small high voltage fet's and a bunch of caps/resistors.

See attached photo .. motor runs well at full torque from 100rpm to max rated (4000 for this motor).
I have not obtained a higher KV motor to test at higher rpm'3 yet ..

The motor can run from 12v to 48v (In fact you can adjust the voltage while the motor is running, and it does not miss a beat, as the sensorless circuit copes with it ..)
[image]

By the way the DC/DC Converter you specified is no longer available. In the picture I have a temporary larger one attached for testing.

bmentink, 2024-06-27:
Update: I have tried my sensorless version on 4 motors now, all different sizes and KV ratings. It all just works, you don't have to calibrate anything, just attach the motor and go ..

bmentink, 2024-07-04:
And here is the adapter board running a RP2040-Zero
[image]

emstraske, 2025-01-16:
Hello, would it be possible for you to share the board files and code for the sensorless version of the controller? I’d love to study and learn from your implementation. However, I understand if you prefer to keep it private.

Thank you for sharing your progress, and I look forward to any insights you're willing to share!